mantidproject/mantid #26414: Workspace/Algorithm Smoke Tests

NickDraper, 2019-07-19:
# Workspace/algorithm smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

* Load some workspaces, try 5 from different instruments
* Show detectors, check detector view GUI, follow non-Python directions [in the documentation](https://www.mantidproject.org/MantidPlot:_Instrument_View)
	 - [ ] Render tab 
	 - [ ] Draw tab 
	 - [ ] Pick tab 
	 - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [the documentation](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms)
	- [ ] Help button 
	- [ ] Validation of inputs 
	- [ ] Overwriting workspace 
	- [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the documentation](https://www.mantidproject.org/MantidPlot:_The_Workspace_Menu#Save_Nexus).

LolloB, 2019-07-19:
Tested on Ubuntu 18.04. No issues found.

alicerussell1, 2019-07-19:
Tested on windows 10, no issues found

Pasarus, 2019-07-22:
Windows 10 again, no issues found.

martyngigg, 2019-07-22:
Tested on macOS. Offline Help not working.

closed 2019-07-22 by NickDraper