mantidproject/mantid #26415: Smoke Testing Umbrella issue

NickDraper, 2019-07-19:
This is to keep track of the coverage of smoke tests during the release cycle.
Please tick and add your name against an item when you are taking it

## Clean/Dirty Install Smoke Tests: #26408
 - [] Windows
 - [ ] OSX
 - [] Linux 

## Fitting Smoke Tests: #26410
 - [ ] Windows
 - [ ] OSX
 - [ ] Linux
 - [] IDAaaS

## Interface Smoke Tests: #26411
 - [ ] Windows
 - [ ] OSX
 - [ ] Linux

## Plotting Smoke Tests: #26412
 - [ ] Windows
 - [ ] OSX
 - [ ] Linux

## Python Smoke Tests: #26413
 - [ ] Windows
 - [] OSX
 - [ ] Linux

## Workspace/Algorithm Smoke Tests: #26414
 - [ ] Windows
 - [ ] OSX 
 - [ ] Linux

martyngigg, 2019-07-19:
Mac is missing the C++ interfaces for workbench. #26420 created.

NickDraper, 2019-07-19:
Reset for the release candidate build on Monday

NickDraper, 2019-07-22:
Reset and cut down for final tests

closed 2019-07-23 by NickDraper