mantidproject/mantid #26419: HFIR open project raises exception then closing dialog window

Harrietbrown, 2019-07-19:
### Expected behavior
When closing load project window in HFIR 4Circle Reduction mantid should not rise an exception.
### Actual behavior
Closing the open project (or any save/load dialog window) HFIR expects to be given a string to load even if open is not clicked
### Steps to reproduce the behavior
1. Open workbench
1. Open Interfaces --> Diffraction --> HFIR 2Circle Reduction
1. go to File --> Open Project
1. close the Choose Project File.
### Platforms affected
Windows.

NickDraper, 2019-07-19:
I've just verified that this was also the case for r4.0 so this is not a step backwards

NickDraper, 2019-07-19:
@peterfpeterson This is an issue we have found during smoke testing that was also present in the last release.  My temptation is to say that we push it out of this release, however if someone at ORNL wanted to fix it, (and it was very low risk to do so). then if you could get a fix in and merged before the builds run tonight it could be included.

wdzhou, 2019-07-19:
@NickDraper @peterfpeterson Till Nov. 3rd, no one in ORNL will use this interface.  So I think it better to put it to the next release.

closed 2021-03-02 by stale[bot]