mantidproject/mantid #26423: Abins: Direct/2D outputs

ajjackson, 2019-07-19:
It is desirable to generate two-dimensional S(Q,ω) maps from ab initio data, and I am looking at how best to bring this to Abins.

Considerable work has been done on this previously and I have initially been working with some vestigal code from Mantid 3.9.2 and also a further-developed folder of code from my predecessor.
However, merging these with the master branch of Mantid is non-trivial and involves some reworking of the Algorithm file especially. Initial efforts also appear to run very slowly, partly due to an iterative masking process that takes a long time (and many calculations) to reach a satisfactory threshold. There is lots of scope for optimisation but I am beginning with an attempt to benchmark/reproduce what was already done before this is complicated further by refactoring.

I intend to develop this on my own Mantid fork but not attempt a PR into the main project until some verification and optimisation has also been done.

A key implementation detail that I am happy to keep for now is the implementation of 2D scans as a separate "instrument" from the default TOSCA 1D spectrum.

NickDraper, 2019-07-25:
@AnthonyLim23 Is there any way to break this down so we don't get a huge PR to review at the end?

ajjackson, 2019-07-25:
As this is a new feature relative to the current master branch, involving some refactoring of what is there, this is not a simple thing to avoid. However I will endeavour to keep this branch focused on the task at hand.

We could have a PR as soon as things appear to be working and before further analysis and optimisation. This seems like a bad idea from a user perspective, even if it is quite convenient from a development perspective. It is possible to hide the functionality from users by removing the necessary instrument from the interface, as was done in the code in 3.9.2... but I assume that was removed because this is _not_ the preferred approach.

One option could be to maintain another branch for refactoring and stylistic changes that are compatible with existing functionality, and PR this more frequently while rebasing the feature development. From my perspective this would probably cost some time and effort when the Git spaghetti goes wrong - but on balance could be less painful than deferring helpful-but-slightly-off-topic refactoring. I'm also unsure of the typical turnaround for merging PRs that don't "do anything".

NickDraper, 2019-07-25:
Smaller pr’s get tested  and merged much faster. So I would suggest you separate the refactoring from the functionality. Refactoring prs get tested just the same as any other, but it is easier to review a or if it is targeted and doing one thing.

Regards,
Nick Draper

ajjackson, 2021-10-01:
An update on this, examining some requirements through discussion with instrument scientists:

Work so far has been based on an "instrument simulation"-esque scan through angles. This leads to various difficulties with sampling efficiency and remapping to 2-D energy-q space. While we do have some excellent technology for dealing with this (i.e. MSlice) it is in practice quite cumbersome to apply these transformations to simulated data, as various parts of Mantid tend to complain about missing XML data related to sample positions etc.

So in order to make some kind of progress and possibly get better results anyway, I will examine the idea of directly working in energy-q space.

- This requires some more refactoring of Abins, which was built on the (now-broken) assumption that energy and q are always related by instrument geometry
- It would still be useful to apply instrument resolution functions. These can be approximated to a f(w), or a f(q,w, Einit, **chopper_settings) can be obtained with Pychop. The latter would be limited to the accessible energy-q window of the instrument, so some kind of masking or extrapolation is called for.
- Approximating to f(w) would apparently be quite an acceptable starting point as q behaviour is dominated by other effects
- Similarly, while a universal interface to Pychop parameters would be nice it is also possible to represent most practical scenarios with a small number of "stock" settings.
- Instrument priority goes MAPS -> MARI -> MERLIN (but it shouldn't be much trouble to have all three from an implementation standpoint)

In the long run it would be very nice to combine these outputs with coherent simulations from Horace/Euphonic. Graphite would be a good test case, having strong coherent and incoherent scattering. (Unfortunately it's quite hard to compute accurate frequencies. How about diamond?)

A major assumption baked into Abins is a lack of q-dependence in the frequencies; instead we effectively sample a full DOS.
- The quality of this assumption really comes down to how low in q we are seeing interesting features. Hydrogen-bonding is an issue, giving interesting low-frequency behaviour with strong intensity.
- A major benefit of this assumption is that it makes overtones feasible to compute!

ajjackson, 2023-04-05:
This was implemented in #33390

closed 2023-04-05 by ajjackson