mantidproject/mantid #26475: Frequency Domain Analysis should set sensible fitting limits

Mantid-Matthew, 2019-07-24:
Currently in the frequency domain analysis GUI the maximum fitting range defaults to 15, this is far to small for the normal frequency ranges which occur. What the best way to determine a default fitting range is should be investigated and then implemented.

AnthonyLim23, 2019-08-14:
We should probably use the context to determine the start and end values. Then we can have a check in the context to see if it is frequency data and if it is return the new fitting range. We would need to be careful to make sure we only use the 'new' range when the data switches between time and frequency data. It should not replace the range if we just change the data (but its from the same domain).

AnthonyLim23, 2019-09-02:
also should put a check in for #25821

closed 2021-03-01 by AnthonyLim23