mantidproject/mantid #26497: Apple Developer registration

rayosborn, 2019-07-26:
Some IT groups refuse to give administrator privilege to those issued with laptops. I currently have a MacBook Pro. Although I can download an application like MantidPlot and install it in my home directory (~/Applications), I cannot run it because my system won't allow software from an unknown developer to be launched. This would require the Mantid Project to obtain an Apple Developer ID certificate, which I think would be justified for a project this size. 

### Platforms affected
MacOS

DanielMurphy22, 2019-12-09:
Although Mantid does not have the Apple Dev ID certificate, MantidPlot (and Workbench) can be opened on Mac: 

**Instructions**
Open the Applications folder in a Finder Window, right-click on Mantid and Select 'Open'.
It will try to verify this and then bring up a pop-up saying it is downloaded from the internet, are you sure you want to open it. 
Clicking Open will run Mantid successfully

Clicking on its Icon in the toolbar or Launchpad should work from now on (until you reinstall mantid!). You will have to do this separately for Plot and Workbench. 

I am closing this issue as this work around is a one-time thing (not everytime you launch Mantid), but of course the Long term goal to obtain a certificate should remain. 

@rayosborn if my instructions on how to open Mantid on Mac above didn't work for you then feel free to message me and reopen the issue!

rayosborn, 2019-12-09:
When I try to open the file in the Finder, I get

[image]

It's true that, when I go to the "Security and Privacy" preference panel, there is a warning message about the application that has a button labelled "Open Anyway". However, when I click on it, I have to authenticate with an admin account username and password, which I don't have. So even on Mac OS 10.14 Mojave, this doesn't work. It is my understanding that it will be even more restrictive in Mac OS 10.15 Catalina, which I cannot test yet. The critical point is that, if I discover I need MantidPlot for the first time when I'm visiting a facility, I will not be able to run it.

NickDraper, 2020-04-03:
Price is apparently 99USD/ year

rayosborn, 2021-02-22:
Since I have not done an experiment at the SNS for over a year, I haven't needed to use MantidPlot, but I have no reason to believe this issue has gone away.

closed 2021-08-29 by stale[bot]