mantidproject/mantid #26513: Workbench: Can't save existing project with new name

PhilColebrooke, 2019-07-29:
### Expected behavior
You should be able to save an existing project with a new name.

### Actual behavior
Workbench closes without saving.

### Steps to reproduce the behavior
1. In Workbench, open an existing project.
2. Make a change to the project, e.g. load another workspace.
3. Click the X in the corner to close Workbench.
4. When asked if you want to save before closing, click Yes.
5. When asked if you would like to overwrite the project. click No.

Workbench also closes if you click the X in the corner of the last dialog, when it would be better if it only closed the dialog and Workbench stayed open.

### Platforms affected
Presumably all.

gemmaguest, 2019-08-02:
There's another related problem: if you get as far as the Save As dialog (either when saving an existing project or a new one) and click Cancel, then Workbench closes. Would it be better if it just closed the dialog without closing Workbench?

PhilColebrooke, 2019-08-02:
> There's another related problem: if you get as far as the Save As dialog (either when saving an existing project or a new one) and click Cancel, then Workbench closes. Would it be better if it just closed the dialog without closing Workbench?

I've also had a go at fixing that, see #26519

PR #26565 (2019-08-15): Workbench: make closing the save and overwrite dialog behave correctly
**Description of work.**
#26516 made it so that when the dialog window asking if you want to overwrite the existing file appeared, clicking the X in the corner would wrongly cause the dialog for choosing file location to appear. This has been fixed by adding a cancel button to the dialog. Some code which appeared to never be executed has also been removed.

**To test:**
1. Open an existing project in Workbench.
2. Make a change, e.g. load a new workspace.
3. Click the X in the top-right corner to close Workbench.
4. When as [... cut]
closed 2019-08-15 by SimonHeybrock