mantidproject/mantid #26533: Add Options dialog for reflectometry GUI

gemmaguest, 2019-07-31:
The ISIS Reflectometry GUI used to have a menu option to open an options dialog. This should be re-added. The dialog allows control of:
- the precision shown in the table
- whether warnings should be shown for:
  - table changes will be lost
  - you're about to process only a subset of a group
  - you're about to process all rows in the table

The warnings above should also be added (the only one currently implemented is to warn on losing table changes when starting autoprocessing, but we should also warn when closing the GUI)

giovannidisiena, 2019-10-09:
Confirmed: AutoProcess **does** ask before overwriting "empty" table.

I have observed that if you pause and then click AutoProcess again it no longer asks, however if you click Search and then AutoProcess the warning reappears. Is this expected behaviour?

giovannidisiena, 2019-10-09:
> Confirmed: AutoProcess **does** ask before overwriting "empty" table in the latest Nightly build.
> 
> I have observed that if you pause and then click AutoProcess again it no longer asks, however if you click Search and then AutoProcess the warning reappears. Is this expected behaviour?

Also note #26981

gemmaguest, 2019-10-09:
When you pause and resume it does not change the table, so this is correct - it's only when you change the search criteria that the table is (or may be) changed, so you get the warning

gemmaguest, 2019-10-25:
Changed milestone to 4.3 because this requires related work to the ISIS Reflectometry GUI which is not implemented yet

PR #27656 (2020-01-23): Add options dialog to Reflectometry GUI [PART 2]
**Description of work.**
This PR is **2** of **6**.

The options dialog view and presenter have been created and integrated into the main window.

**To test:**
Complete a code review and check that the dialog is displayed when requested. Full manual testing instructions will follow when subsequent PRs are merged.

Partially resolves #26533

*This does not require release notes* because **they will be added in a subsequent PR**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/Reviewi [... cut]
PR #27657 (2020-02-07): Add options dialog to Reflectometry GUI [PART 3]
**Description of work.**
This PR is **3** of **6**.

The warning for discarding changes has been implemented.

**To test:**
Complete a code review and check that the warning is displayed when expected (e.g. editing and closing a batch, loading a new batch on top of existing changes, etc) and the negative case (e.g. warning should not be displayed if a batch is first saved, then closed/overwritten). Full manual testing instructions will follow when subsequent PRs are merged.

Partially resolves #26533

*This does not require release notes* becaus [... cut]
PR #27658 (2020-02-21): Add options dialog to Reflectometry GUI [PART 4]
**Description of work.**
This PR is **4** of **6**.

The warning for processing all groups or partial groups has been implemented.

**To test:**
Complete a code review and check that the warning is displayed when expected (e.g. no rows selected, all groups with content selected, all rows selected, etc). Full manual testing instructions will follow when subsequent PRs are merged.

Partially resolves #26533

*This does not require release notes* because **they will be added in a subsequent PR**

---

#### Reviewer ####

Please comment on the follo [... cut]
PR #27660 (2020-02-21): Add options dialog to Reflectometry GUI [PART 5]
**Description of work.**
This PR is **5** of **6**.

The RoundPrecision widget has been implemented.

**To test:**
Complete a code review and check that the table precision is displayed as expected (e.g. default value if unchecked, user-specified value if checked, default value if unchecked again, etc). Full manual testing instructions will follow when subsequent PRs are merged.

Partially resolves #26533

*This does not require release notes* because **they will be added in a subsequent PR**

---

#### Reviewer ####

Please comment on the follo [... cut]
closed 2020-05-12 by martyngigg