mantidproject/mantid #26598: Muon Analysis 2 Workflow improvements umbrella issue

Mantid-Matthew, 2019-08-08:
The two main workflows for the Muon Analysis GUI are summerised below along with the suggested changes in order improve how well the GUI matches them.

Transverse Field Mode
==================
In transverse field mode there are a number of groups and no asymmetry pairs. In this mode the most common fit to do is a simultaneous fit over all the groups in the experiment. Functionality missing from this workflow is:
  * The ability to sequentially do a simultaneous fit over several runs. In general it would be good if sequential fit could be changed so that you can specify a fit template to then run over multiple runs.
  * When transverse field data is loaded in the correct graphs to be shown are a tiled plot of all the groups, likewise the correct default fit selection is a simultaneous fit over all the groups.

AnthonyLim23, 2019-08-14:
If we are doing a sequential simultaneous fit, would it be fair to assume we will never use different runs in a single simultaneous fit? This could make it easier for us to implement.

Mantid-Matthew, 2019-08-14:
In the standard use case you do a simultaneous fit over all the groups of one run. There may be edge cases where you want to do simultaneously fit over multiple runs but I think we should concentrate on the common workflows first.

AnthonyLim23, 2019-09-17:
@ciaranightingale I think the work we have discussed will fix this

closed 2020-09-07 by AnthonyLim23