mantidproject/mantid #26602: FDA plot type FFT All causes crashes

Harrietbrown, 2019-08-09:
### Expected behavior
When changing plot type on frequency domain analysis to `Frequency FFT All` the plot should be updated propperly. 
### Actual behavior
When changing plot type to FFT All when multiple runs are loaded FDA raises an error
### Steps to reproduce the behavior
1. open workbench
1. open Frequency domian analysis
1. load runs 62260-62261
1. go to the transform tab.
1. turn off `Imaginary Data` and then calculate FFT
1. go to home tab, and change plot type to `Frequency FFT All`
1. an error is raised.
### Platforms affected
tested on windows

AnthonyLim23, 2019-08-09:
ah ok, I think I know what this is

AnthonyLim23, 2019-08-09:
I just had a go on my nightly and it worked. On master I have a bug, @Mantid-Matthew can you reproduce this in nightly or master?

PR #26606 (2019-08-09): Fix FDA error when no Im data and phaseQuad plot
The FDA GUI had a couple of bugs with plotting, this fixes them.

**To test:**
test 1:
open workbench
open Frequency domian analysis
load runs 62260-62261
go to the transform tab.
turn off Imaginary Data and then calculate FFT
go to home tab, and change plot type to Frequency FFT All

test 2
Go to the phaseQuad tab and generate some data
Go to the transform tab and do some FFT's of phasequad
Go to home tab and try the various FFT plot options
The phaseQuad data should show up

Fixes #26602 . 

---
In release notes
#### Reviewer ####

Please comm [... cut]
closed 2019-08-09 by martyngigg