mantidproject/mantid #26613: Remove or fix drag and drop tabs from Muon Analysis

Mantid-Matthew, 2019-08-09:
Currently the code to allow tabs in muon analysis to be dragged and dropped back into the main GUI is disabled as it does nothing on Linux and windows and leads to errors on OSX. This should either be fixed to work correctly or removed.

AnthonyLim23, 2019-08-14:
@Mantid-Matthew is this done?

Mantid-Matthew, 2019-08-14:
No this is an issue to properly resolve something that was quickly fixed previously.

Mantid-Matthew, 2020-09-01:
This is related to #24678 as this issue comes back to the fact that the tabbing on the new Muon Analysis GUI is not well implemented. This leads to numerous usability issue particularly on OSX and I suspect if any of the instruments scientists used anything except windows it would be receiving substantially higher priority. We are currently using a handrolled tabbing implementation but would be much better served to use the QMainWindow QDockWidget implementation which comes with Qt and would resolve many of these issues.

AnthonyLim23, 2021-08-25:
@DavidFair does the tabbing work ok on mac now (you need to double click to undock/redock)?

sf1919, 2021-11-25:
This is now blocked by #33044

ConorMFinn, 2022-02-03:
Popping out of tabs is now performed by double-clicking them. Dragging code removed via #33141 due to segfaults on macOS. Windows are movable above the main analysis GUI. Found a bug with the GUI re-attaching the windowed tabs but only on macOS, issue at #xxxxx

closed 2022-02-03 by ConorMFinn