mantidproject/mantid #26632: Settings window - handle CONF.get returning unexpected type

DTasev, 2019-08-13:
[mantidworkbench_ini.zip](https://github.com/mantidproject/mantid/files/3496032/mantidworkbench_ini.zip)

### Expected behavior
Opening the settings window should handle unexpected types for `prompt_save_on_close` and `prompt_save_editor_modified`. 

### Actual behavior
For some reason the two settings above can become `string`s instead of `bool`s. This throws an unhandled exception.

### Steps to reproduce the behavior
- Replace the `~/.config/mantidproject/mantidworkbench.ini` (Ubuntu) `~/AppData/Local/mantidproject/mantidworkbench.ini` (Windows) file with the one uploaded in this PR. 
    - Make sure to backup your original one if you don't want to lose your settings.
- Open the Workbench and then the File > Settings.

### Platforms affected
Possibly all

hsautessella, 2019-09-03:
To go along with this issue, it appears that the `Editors/SessionTab` key in the options can get itself in to a state where it cannot be read This causes Workbench to crash on startup, with no way for the user to reset the settings unless they directly edit the settings file.

 I've not been able to consistently recreate it, but below I've got an invalid value for the key.
```
[Editors]
SessionTabs=@Variant(\0\0\0\x7f\0\0\0\xePyQt_PyObject\0\0\0\0\0)
```
At the moment, the session tabs are saved as a set, which may be the root of the problem. Maybe we should save them as a list instead. Note, we do not want to save duplicate tabs #25004. It may be nice to preserve the order of the tabs as well, see [here](https://stackoverflow.com/questions/480214/how-do-you-remove-duplicates-from-a-list-whilst-preserving-order) for tips on how to do this.

martyngigg, 2019-11-13:
When reporting #27363 James found he couldn't open workbench after the crash. The `.ini`  file contains:

```
[Editors]
SessionTabs=@Variant(\0\0\0\x7f\0\0\0\xePyQt_PyObject\0\0\0\0\xf3\x63__builtin__\nset\np0\n((lp1\nVC:/Users/jsl28/Desktop/test_py23.py\np2\naVM:/doc/Summer student 2019/Doug internship work/Python/Degrader Calculator.py\np3\naVM:/doc/Summer student 2019/Doug internship work/Python/degrader_dictionary2.py\np4\natp5\nRp6\n.)
```

and this looks dodgy. It might mean we at least have a way to reproduce it.

PR #27853 (2020-02-10): Added some protection around loading config files in workbench
**Description of work.**

Handles corrupt QObjects, and some of the sensitive fields in MessageDisplay, that HAVE to be boolean as they are passed to a C++ base class

**To test:**

1. Find your mantidworkbench.ini file (on windows it is in %appdata%\mantidproject)
1. Take a copy of your file so you can restore it after
1. Delete it
1. Start workbench, if it creates a new .ini file you know you re looking at the right file
1. Stop workbench and edit the new file. Paste in this corrupted version.
```
[General]
high_dpi_scaling=true
 [... cut]
closed 2020-02-10 by gvardany