mantidproject/mantid #26675: Speed issues with TOSCA reduction on workbench

martyngigg, 2019-08-21:
**Original reporter:** Forum user: https://forum.mantidproject.org/t/mantid-4-0-workbench/463/17

### Expected behavior

Reduction of data Indirect data in MantidPlot/Workbench should be the same. 

### Actual behavior

Reducing TOSCA data:

    A simple data reduction (multiple run numbers, summed, default rebin) seems to take MUCH longer than it used to. Unsure if this is expected.

DanielMurphy22, 2020-02-25:
Had this reported again by Hamish. I don't see this getting done for the release so moving to next release so it won't be forgotten.

martyngigg, 2020-02-25:
@DanielMurphy22 I'd put this in `5.1S1` or it'll get left until the final 5.1. We might need to work between core and spectroscopy to figure this one out as I suspect it might be more of a workbench issue if the same workflow works in MantidPlot properly.

Mantid-Matthew, 2020-04-06:
@martyngigg Do you think this is resolved by the various workbench speed fixes that have been put in?

martyngigg, 2020-04-06:
@Mantid-Matthew I'm not sure I am best placed to confirm this as I didn't report the original issues but I do agree that ll of the fixes will certainly help. Perhaps we need to get a build into the hands of a scientist that reported it and see what they think?

DanielMurphy22, 2020-04-06:
I'm sure Hamish would be willing to test this.

Mantid-Matthew, 2020-04-06:
@DanielMurphy22 I think all of the relevant PR's should be in nightly now. If you could contact Hamish about giving it a re-test that would be great.

AnthonyLim23, 2020-04-14:
@DanielMurphy22 did you find out if this had been fixed?

DanielMurphy22, 2020-04-14:
Not been tested by Hamish yet. I can try myself.

AnthonyLim23, 2020-04-14:
that would be great thank you. Do you know how slow it use to be, so we know if it has got faster or not?

DanielMurphy22, 2020-04-14:
No but I've got 4.2 and 4.1 installed so can test them!

DanielMurphy22, 2020-04-15:
Hamish has tested and quote: "they seemed identical, which is a great improvement for the workbench"

When I performed my independent testing, I found these numbers:

|      Version     | Workbench / s | Plot / s |
|:----------------:|:-------------:|:--------:|
|        4.2       |      57.0     |    3.5   |
|        5.0       |      12.2     |    3.7   |
| Nightly 07042020 |      3.9      |    3.8   | 

Well Done!

AnthonyLim23, 2020-04-15:
That makes me very happy! 

I will close the issue

closed 2020-04-15 by AnthonyLim23