mantidproject/mantid #26688: Add Placzek correction

martyngigg, 2019-08-22:
After #26454 work has been completed the full Placzek correction algorithm needs to be implemented.

Confirm with Marshall the exact steps required for the correction and implement.

marshallmcdonnell, 2019-08-22:
The code to implement the "basic" Placzek correction with input L1, L2s, ant two theta angles as lists required for the correction are here:

marshallmcdonnell, 2019-08-22:
The references are buried in the code comments (here)[https://github.com/marshallmcdonnell/mantid_total_scattering/blob/9f762d9e72149bcc87c5d1958aff83aa4c5c2c61/total_scattering/inelastic/placzek.py#L151-L157] for the articles that this implementation is based on.

martyngigg, 2019-08-23:
Thanks Marshall! This is for us to get started.

closed 2019-09-25 by DanNixon