mantidproject/mantid #26692: Use 1:1 mapping for det to spectra in KafkaStreamDecoders.

LamarMoore, 2019-08-22:
Currently the streamers expect to receive a stream for the detector to spectrum mapping which is a purely ISIS concept and will not be utilized by the ESS. This needs to be accounted for in the stream decoders.

OwenArnold, 2019-08-22:
@LamarMoore can you provide any more detailed information around this issue please. I'd like to find someone to pick it up.

OwenArnold, 2019-08-22:
@LamarMoore could you also describe what the current work arounds would look like, so that we know how painful this would be to work with at v20 in current state.

matthew-d-jones, 2019-08-22:
We would need to send a detector-spectrum map via Kafka along with the run start message. We don't have a workaround in place, we would probably have to add some code to Nicos to serialise this extra message.

If I remember correctly from our previous conversations about this, generating a 1:1 map would be consistent with what happens when loading non-ISIS event data from file.

commit 52414d55 (2019-09-20): Create 1:1 spec-det mapp when message is missing

When subscription to the topic that provides the spectrum to detector ID
mapping fails or the message is otherwise not consumed generate a 1:1
mapping by loading the instrument before creating the buffers.

This is a slightly hacky solution in time for V20 beam time, the
spectrum to detector ID mapping may be later moved to the run start
message, making this easier to handle.

Fixes #26692
commit 445e41cc (2019-11-19): Create 1:1 spec-det mapp when message is missing

When subscription to the topic that provides the spectrum to detector ID
mapping fails or the message is otherwise not consumed generate a 1:1
mapping by loading the instrument before creating the buffers.

This is a slightly hacky solution in time for V20 beam time, the
spectrum to detector ID mapping may be later moved to the run start
message, making this easier to handle.

Fixes #26692
closed 2019-09-20 by martyngigg