mantidproject/mantid #26740: Error getting states for LARMOR reduction

gemmaguest, 2019-09-02:
It has been reported that every reduction on LARMOR is reporting the following errors:
```
Errors in getting states...
StateMaskDetector: The provided inputs are illegal. Please see: {"Missing detector name.":"detector_name: None\nMake sure that the detector names are set.", "Missing short detector name.":"detector_name_short: None\nMake sure that the short detector names are set."}
```

This has also been seen occasionally whilst reducing SANS2D data, but has not been seen on LOQ.

We should ask Rob if this is still a problem and for an example user file and run for LOQ (and SANS2D if they can provide one).

gemmaguest, 2019-09-27:
Removed milestone as we are waiting for scientist input on this

smk78, 2020-07-16:
Rob D tested this in Workbench on IDAaaS today. He says it seems to be fixed so I am closing the ticket.

closed 2020-07-16 by smk78