mantidproject/mantid #26767: V20 Master Issue for Cleanup from V20

LamarMoore, 2019-09-06:
- [ ] Need to make topics more generic. I.e. take lists of different types of topics and connect to all topics provided by `Facilities.xml` file
- [ ] The multi-threading (partitioning) of the event buffer in the `KafkaStreamDecoder` is broken in the `computeGroupBoundaries`.
- [ ]

OwenArnold, 2019-09-09:
Thanks for recording @LamarMoore. Can we look into creating issues for each of these at a time that suits you in the next few weeks.

OwenArnold, 2019-09-18:
@LamarMoore, your last point ```There seems to be an issue with the SaveNexusESS algorithm not saving monitor transformations```. I have not tried saving/reloading with some of the V20 files that I have. Can you either specify the runs that were problematic so that I can retry, or try it yourself based on `master` which now contains a few fixes relevant.

closed 2019-11-29 by DanNixon