mantidproject/mantid #26778: Live data continues infinitely if ISISReflectometry GUI closed without stopping

giovannidisiena, 2019-09-11:
# Expected behavior
When the `ISIS Reflectometry` GUI is closed live data monitoring should stop, or there should be a prompt on clicking the close button.

Discovered during #26466 development, refs #22051.
### Actual behavior
The GUI closes and live data monitoring continues. Opening the GUI again does not solve this.
### Steps to reproduce the behavior
- Open the `ISIS Reflectometry` interface
- Start live data monitoring
- Wait to observe something similar to `Loading live data chunk 0 at 14:27:50`
- Close the GUI
- Reopen the GUI and observe inability to stop live data monitoring
### Platforms affected
Observed on Windows 10

gemmaguest, 2019-09-19:
This is actually a "feature" which is explained in the [docs](https://docs.mantidproject.org/nightly/interfaces/ISIS%20Reflectometry.html#live-data-monitoring). The algorithm can be stopped from the Algorithm Progress dialog. However, it should be easy to stop it and I agree this would be more intuitive (or we could perhaps at least offer this as an option) so let's discuss this with the scientists.

gemmaguest, 2020-07-20:
Closing because this is low priority and we are unlikely to get around to it. It was a developer suggestion rather than a request from users.

closed 2020-07-20 by gemmaguest