mantidproject/mantid #26805: Investigation needed

OwenArnold, 2019-09-16:
This line looks like a bug to me https://github.com/mantidproject/mantid/blob/master/Framework/NexusGeometry/src/NexusGeometrySave.cpp#L408. Am I mistaken? Shouldn't this be 3. Not sure why we don't get test failures around this?

Could you please investigate or explain?

TakudzwaMakoni, 2019-09-16:
if my understannding is correct, the dataset contains the magnitude of the vector. the unit vector is stored as an attribute `vector`?

https://github.com/mantidproject/mantid/blob/2ffc539e39bb827978cf81a20fdff9af21f57ecc/Framework/NexusGeometry/src/NexusGeometrySave.cpp#L423

TakudzwaMakoni, 2019-09-16:
maybe something to do with static casting hsize_t? i have had compiler warnings relating to that before

OwenArnold, 2019-09-16:
Correct

closed 2019-09-16 by OwenArnold