mantidproject/mantid #26837: Create set of functions to plot satellite peaks for easier determination of modulation vectors

martyngigg, 2019-09-19:
As a user I want to be able to determine the modulation vectors for a given set of peaks.

The SNS has a script for creating a plot of the satellites and finding clusters of satellites by collapsing outer peaks onto a smaller HKL subspace. This provides a visual aid for the scientist to calculate the modulation vectors.

Take the script, clean it up and make it work for testing on WISH.

RichardWaiteSTFC, 2019-12-02:
The script (attached) works on WISH data (input an ISawPeaks table and UB matrix). 
Ideally we would like to use FindSXPeaks to generate the peak table but it is difficult to pick up enough weak peaks. An intial workflow might be to PredictFractionalPeaks with the rough modulation vector and then CentroidPeaksMD. 

The detector coverage on WISH is not a good as TOPAZ so there needs to be a way of combining different runs - this may be as simple as combining peak workspaces.

[TOPAZ_8071_8x8x5_sat.zip](https://github.com/mantidproject/mantid/files/3912085/TOPAZ_8071_8x8x5_sat.zip)

RichardWaiteSTFC, 2019-12-09:
Tested again on face-centred Ni2MnGa data with satellite at (0.4,0.4,0) and second-order reflection at (0.8,0.8). The script assumes Bragg peaks are at all integer HKL and so doesn't account for the centring (see screenshot below)
 
[image]

Talked to Pas on WISH and will need an additional step to convert to the correct primitive cell - but this will be difficult to automate. Also WISH users would like to be able to determine which reflection corresponds to the first-order reflection as distinct from higher order reflections. This would require integration before running the script.

DannyHindson, 2019-12-16:
Create a separate issue to think about workflow

closed 2019-12-16 by DannyHindson