mantidproject/mantid #26841: Bug in parsing NeXus geometry meshes

matthew-d-jones, 2019-09-19:
There is at least one bug around here:
https://github.com/mantidproject/mantid/blob/5fa751ee1434a83c200cfc777f4b253dbaccc0f1/Framework/NexusGeometry/src/NexusGeometryParser.cpp#L461

Code seems to assume that all faces in the mesh are mapped to a detector ID, this does not have to be the case.
It also looks like it may assume that the number of vertices is the same for all faces in the mesh, this is also not necessarily the case.

OwenArnold, 2019-09-19:
@matthew-d-jones do you have particular test files to exercise these problems?

matthew-d-jones, 2019-09-19:
Example file is on the network drive at `olympic/Babylon5/Public/Matthew Jones/DETGEOM_example_bug.nxs`

PR #26894 (2019-10-01): Shape parsing fixes
Fixes #26841

Specifically addresses:
1. Bug: all faces = detector face assumption in "detector_shape" OFF meshes
1. Missing feature: No handling for "detector_shape" and cylinders
1. Bug: Tube optimisation dispensing with detectors falling outside tubes
1. Feature: Better ability to write code at the `NX` level with basic data types. i.e. `readNXInt` with less concern with underlying type and coercion, also provides good error reporting for cases that cannot be handled, which is critical for fast fixes on beamlines.
1. Extra testing written for all above cases.
1. Other cl [... cut]
closed 2019-10-01 by DanNixon