mantidproject/mantid #26904: ALFView renaming

AnthonyLim23, 2019-09-27:
At present ALFView's code does not always follow the coding standard. This issue is to fix that. However, we will need to first determine if the GUI is useful to any other areas, as this may alter our choice of name.

AnthonyLim23, 2019-11-28:
need to rename:
ALFView -> ALFInstrumentViewer
ALFView_view -> ALFInstrumentView
ALFView_model ->ALFInstrumentModel
ALFView_presenter ->ALFInstrumentPresenter

need to move the following to https://github.com/mantidproject/mantid/tree/429bd5b007b3639268cd77f3b0535f75cc9a76f0/qt/widgets/common/inc/MantidQtWidgets/Common
BaseInstrumentModel
BaseInstrumentPresenter
BaseInstrumentView
PlotFitAnalysisPaneModel
PlotFitAnalysisPaneView
PlotFitAnalysisPanePresenter

AnthonyLim23, 2019-11-28:
@DanNixon does this seem like a better set of names/locations for these files?

DanNixon, 2019-11-29:
I'd keep the full GUI name in the component names, i.e.
ALFView_view -> ALFInstrumentViewerView
ALFView_model ->ALFInstrumentViewerModel
ALFView_presenter ->ALFInstrumentViewerPresenter

AnthonyLim23, 2019-12-02:
I was trying to avoid things such as ViewerView as it looks confusing

DanNixon, 2019-12-02:
> I was trying to avoid things such as ViewerView as it looks confusing

Breaking a naming convention is in itself confusing.

AnthonyLim23, 2019-12-02:
true.

PR #27577 (2020-01-08): Rename ALFView files and restructure
**Description of work.**
The file and varaible names for ALF view were not very good. This PR improves them, they now follow the coding standards. I have also moved the common functionality to a more readily available part of Mantid. 

**To test:**
Open Mantid
Go to custom interfaces -> Direct -> ALF view
An interface should apper
In the run box type `80511` and press enter
Wait a little while and an updated instrument should appear in the GUI
Go to the pick tab
Select the `tube` option
Change the subplot from the pick tab to integrate (right click)
Change  [... cut]
closed 2020-01-08 by SimonHeybrock