mantidproject/mantid #26924: Make IPython custom shell compatible with IPython >= 7.1

jmborr, 2019-10-01:
### Actual behavior:
Any command in the IPython shell results in the following error:
```
TypeError                                 Traceback (most recent call last)
TypeError: our_run_code() got an unexpected keyword argument 'async_'
```

### Cause:
[IPython.core.interactiveshell.run_code()](https://ipython.readthedocs.io/en/7.1.0/api/generated/IPython.core.interactiveshell.html?highlight=run_code#IPython.core.interactiveshell.InteractiveShell.run_code) for IPython >= 7.1 introduces keyword argurment `async_=False` which is not taken into account in the custom implementation [our_run_code()](https://github.com/mantidproject/mantid/blob/master/MantidPlot/ipython_widget/mantid_ipython_widget.py#L39) of this function.

jmborr, 2019-10-01:
@martyngigg Hi Martin,

Turns out that the ipython widget of MantidPlot and the Workbench breaks for IPython >= 7.1 because method `run_code` has become an `asyncio` coroutine.

This commit seems to fix the issue for MantidPlot but not for the Workbench. I'm quite ignorant of `asyncio` so I'd appreciate it if you can give me a hand with this.

martyngigg, 2019-10-02:
Hi Jose. I've just tested out your fix on my machine with Python 3/IPython 7.8 and it seems to execute correctly in both MantidPlot and Workbench. What error are you seeing in Workbench?

jmborr, 2019-10-02:
See the below screenshot (please ignore the HDF5 related error). It only happens on entering the first command, it is initializing some object. After that there's no more error messages.

I'm using Python 3/IPython 7.7.0

[image]

martyngigg, 2019-10-03:
I think this is a warning that can be safely ignored and has been fixed in a later version of `qtconsole`: https://github.com/jupyter/qtconsole/pull/313. I was able to replicate your error inside a virtualenv and once i had done `pip3 install -U qtconsole` the error went away.
Are you able to upgrade `qtconsole`?

jmborr, 2019-10-10:
> I think this is a warning that can be safely ignored and has been fixed in a later version of `qtconsole`: [jupyter/qtconsole#313](https://github.com/jupyter/qtconsole/pull/313). I was able to replicate your error inside a virtualenv and once i had done `pip3 install -U qtconsole` the error went away.
> Are you able to upgrade `qtconsole`?

Hi Martyn,
Sorry for the very late reply, very busy with other things but yes, the warning has gone away after upgrading `qtconsole`.

martyngigg, 2019-10-11:
Hi Jose. I think this means that the PR is okay to go?

jmborr, 2019-10-11:
Yes, but the two Ubuntu build servers continue to fail with what it looks like an unrelated problem:

```
stderr: fatal: unable to access 'https://github.com/mantidproject/mantid.git/': Received HTTP code 503 from proxy after CONNECT
```
It's up to you if you still think it's OK to approve this PR

martyngigg, 2019-10-11:
Have you fixed the Python 2 issue?

jmborr, 2019-10-11:
> Have you fixed the Python 2 issue?

oohhh you're right, I forgot! I'll get down to it

closed 2019-10-14 by DanNixon