mantidproject/mantid #26935: User defined peak height on fitting GUI not accounting for normalisation by bin width on Mantid Workbench only

RichardWaiteSTFC, 2019-10-02:
### Expected behavior
When a user adds a peak to fit to a plot with the GUI, the height returned to the function should by the ymax*bin_width (as the units of the y-axis of the plot are counts/bin_width e.g. counts/meV)
 
### Actual behavior
In mantid workbench the normalisation of the counts by the bin width is not accounted for and therefore the initial guess for the height is different from expected.

### Steps to reproduce the behavior
Fit any peak (e.g. gaussian) to data in a workspace, select the fitting tab, right-click to add a peak, choose height, fwhm etc. then plot initial guess.

[image]
[image]

### Platforms affected
mantid workbench

NickDraper, 2019-10-25:
Daniel please check if this is in the current version

DanielMurphy22, 2019-11-22:
In Mantid WB v4.2 this seems to work correctly!

closed 2019-11-22 by DanielMurphy22