mantidproject/mantid #26943: Seperate the ISIS and Mantid YUM repo

NickDraper, 2019-10-03:
For release 4.3 we need to delay the release to ISIS by a couple of weeks after the Mantid release, otherwise all the iDAaaS machines will update during the cycle.

The idea is to create a new YUM repo perhaps called Mantid and move the automatic releases and download page to use that.

Then the ISIS repo will be used whenever we choose to deploy to ISIS, in a similar way that the SNS does.

NickDraper, 2019-10-03:
@ElliotAOram Does this work with or have an impact on Autoreduction?

martyngigg, 2020-02-25:
Rather than maintain two separate repos we have requested that auto updates of the main mantid release to be disabled on IDAaS and the cabin machines at ISIS.

closed 2020-02-25 by martyngigg