mantidproject/mantid #26981: ISIS Reflectometry interface hangs when Mantid is closed

giovannidisiena, 2019-10-09:
### Expected behavior
ISIS Reflectometry interface should close when Mantid is closed

### Actual behavior
ISIS Reflectometry interface hangs after Mantid is closed

### Steps to reproduce the behavior
- Open ISIS Reflectometry interface
- Enter investigation id: 13460
- Search
- Autoprocess
- Select 'no' on warning
- Close Mantid
- Observe detached interface
- Click process
- Observe close (with error report)

### Platforms affected
Windows 10 & Ubuntu 18.04, possibly all

gemmaguest, 2019-10-25:
I think this has the same cause as #26691 so I'll close this and reference it from that issue

giovannidisiena, 2019-10-28:
> I think this has the same cause as #26691 so I'll close this and reference it from that issue

It is also possible to open multiple reflectometry interfaces - is this something worth looking into?

gemmaguest, 2019-10-28:
> > I think this has the same cause as #26691 so I'll close this and reference it from that issue
> 
> It is also possible to open multiple reflectometry interfaces - is this something worth looking into?

It would be good to disallow multiple instances. In theory it should be ok but it's not tested or supported so you could run into problems e.g. autoprocessing the same thing twice or running with different instruments. It's not something users are likely to do, though, so it's probably not worth spending a lot of time on. Could you raise an issue to look into it? If it's a quick fix then it would be good to do it.

gemmaguest, 2019-10-29:
Reopening because following PR #27103 the related issue is fixed but it is still possible to close MantidWorkbench without the reflectometry interface closing. To reproduce it now though you need to start a successful autoreduction:
- instrument INTER and ID 1120015
- click Autoprocess and log in to ICat if necessary
- while runs are being processed, close MantidWorkbench
- observe that the reflectometry interface stays open and still cannot be closed
- if you click Pause then you can now close the reflectometry interface

gemmaguest, 2019-11-18:
Note that in MantidPlot, you get a segfault instead of the interface remaining open.

ConorMFinn, 2021-11-01:
The GUI no longer stays open, but it does ignore the cancel request if workbench is closed. This results in a segfault and the error reporter appearing. This appears to be an issue with a number of other algorithms which react poorly to workbench being closed while they are running. There is no simple fix, and since this is a minor issue, it can be closed for now.

closed 2021-11-01 by ConorMFinn