mantidproject/mantid #26995: ExtractFFTSpectrum missing Properties from FFT

DanielMurphy22, 2019-10-10:
Some issues raised by **James Lord** using Workbench (4.1), through Mantid.help on 08/10/19
(these issues seem to affect both Workbench and Plot)

- ExtractFFTSpectrum needs to make the “Shift and “AutoShift” options available, as the underlying FFT does. Otherwise the Real and Imaginary spectra are useless. Amplitude is OK. Likewise it needs “AcceptXRoundingErrors” if the data comes from a DAE / raw file with single precision bin sizes.

| I can only assume he knows what is best here

- ExtractFFTSpectrum with only a real input and requesting spectra 0-2 (unsigned frequency) gives an output with blank space at the end, left over from the way FFT generates signed and unsigned spectra. It should cut this off.

| I can also recreate this: Blank Space Output at end of Y values in ExtractFFTSpectrum with no Imaginary input

- ExtractFFTSpectrum labels its X axis as “ns” while FFT correctly sees that a muon time domain workspace is in microseconds so labels the output as MHz (whether I select Forward or Backward).

| I can recreate this: In my case I get Time(ns) when FFT gives d-spacing / Angstroms

AnthonyLim23, 2020-07-29:
@alicerussell1 how long do you think these fixes would take?

alicerussell1, 2020-07-29:
They shouldn't take too long - a day at most. 
- The first one should just require declaring the properties and passing them through to the child algorithm.
- The second one should just be adding a check and after the child algorithm has finished which removes the blank output if necessary
- The third one will involve getting the unit from the output workspace and setting it to the workspace, but this is done in  - I think it currently is just hard coded.

PR #29201 (2020-08-27): Improving ExtractFFTSpectrum Algorithm
Made the following improvements to ExtractFFTSpectrum
1. Added properties `Shift`, `Autoshift`, and `AcceptXRoundingErrors`. These are properties of FFT that were apparently missing according to the reporter of the issue.
2. On only real inputs, ExtractFFTSpectrum left a 'tail' of zeros on each histogram, due to the way FFT deals with this. These now get chopped off such that no real data is lost.
3. ExtractFFTSpectrum now returns the units decided by FFT. 

**To test:**

_New Properties_
- For the properties, ensure that enabling each new property gives  [... cut]
closed 2020-08-27 by martyngigg