mantidproject/mantid #27018: 'import nxs' broken on Windows installation

martyngigg, 2019-10-15:
### Expected behavior
`import nxs` is successful

### Actual behavior

The import results in 

```
import nxs as nxs
 
OSError: File C:/Jenkins/workspace/release_clean-win7/external/src/ThirdParty/bin/libNeXus-0.dll from environment variable NEXUSLIB does exist
```

### Steps to reproduce the behavior
Start MantidPlot or Workbench and run `import nxs`

### Platforms affected
Windows with an installed package. A developer build is fine.

DannyHindson, 2019-11-01:
Error was in release 4.1.0 also

Seems to be caused by a failed attempt to overwrite C:\MantidInstall\bin\mantid\kernel\packagesetup.py (which has a full path to the nexus dll) with packagesetup.install.py (which has a relative path) during the install

The overwrite is attempted using a call to mtd_install_files in the following file:

D:\MantidSource\mantid\Framework\PythonInterface\mantid\kernel\CMakeLists.txt

There are only 2 uses of the mtd_install_files function in the whole install so possibly a problem with the function. I think the file (INSTALL ) command it uses perhaps only overwrites files if the datetime stamp on the file being copied is newer than the one that's there already

Will try to debug by adding some "message(STATUS )" statements to the cmake files

martyngigg, 2019-11-01:
@DannyHindson Ah yes I see what is happening now. I wonder if inside `mtd_install_files` we could `touch` the packagesetup.install.py before we try to "install" it to trick the function. We have to use `execute_process(${CMAKE_EXECUTABLE} -E touch` as `file(TOUCH` was added in more a recent version of cmake than we are using.

PR #27249 (2019-11-05): import nxs not working on Windows
**Description of work.**

The "import nxs" command wasn't working due to the NEXUSLIB environment variable containing an absolute path that was specific to the build machine ( eg C:/Jenkins/workspace/release_clean-win7/external/src/ThirdParty/bin/libNeXus-0.dll) rather than a relative path

The cause of this is a file copy failing during the construction of the install package. A file called packagesetup.py is supposed to be overwritten by a different version of the same file containing a relative path for the library but this fails to happen.

The fix adds an [... cut]
closed 2019-11-05 by martyngigg