mantidproject/mantid #27019: "from __future__ import" in Workbench doesn't work!

DanielMurphy22, 2019-10-16:
James Lord has been testing Workbench Mantid 4.1 functionality.

He has tested from __future__ import (print_function, division, unicode_literals) 

and as these 3 do not work correctly, it seems that 'absolute_import' is likely not to work either.

See below for details:

1.	The “from future import print_function” for Python 3 preparations isn’t working as advertised. If I use print (“the answer is”,x) I get a tuple in the messages window such as (‘the answer is’,42.0). It’s as if the future statement is used for syntax checking, rejecting bare print statements, but then turns itself off.

***YES Works in Plot, **not** in Workbench***

2.	To follow up, this works under standard Python 2.x, even using …/MantidInstall/bin/python.exe, but NOT in Workbench (gives a syntax error):
from __future__ import print_function
myprint=print
myprint ( "the answer is", 42 )

***YES Works in Plot, **not** in Workbench***

3.	And it appears the other __ future__ statements are broken too.
“division” doesn’t make 1/3 return 0.33333 (it’s still integer 0 as in plain Python 2.7).

***YES Works in Plot, **not** in Workbench***

4.	unicode_literals is broken too. String literals are not Unicode unless I add the “u” prefix. If I attempt to save a script with a non-ascii character in a string literal, such as £ (with or without a source code encoding indicator as in PEP263) I get an error traceback indicating it was trying to use ascii encoding and the error reporting dialog. And if I put in the encoding indicator and then try to run the (unsaved) script I get a syntax error.

***YES save in Workbench throws up an Error report! Mostly only seems a problem when saving!***

AnthonyLim23, 2019-10-16:
I mentioned this to @hsautessella the other day, Im not sure if he had a chance to look into it.

DavidFair, 2019-10-16:
So print correctly rejects kwarg usage (e.g. `print bar`) however the future imports are definitely not working

Here is a reproducer

``` 
# The following line helps with future compatibility with Python 3
# print must now be used as a function, e.g print('Hello','World')
from __future__ import (absolute_import, division, print_function, unicode_literals)

if (1/3) == 0:
    print("Broken")
else:
    print("Fixed if no brackets printed")
```

I also think we need a Python 3 build too. As there is this assumption that by using future imports Mantid (on Python 2) will behave identically to a Python 3 build. Whilst we hope this is true there may be other subtle differences not covered by future imports too.

hsautessella, 2019-10-17:
I've been away the last couple of days so haven't looked at this yet. I should be able to have a look next week

PR #27073 (2019-10-24): Fix __future__ import behaviour in Workbench scripts
**Description of work.**
The \_\_future\_\_ imports were not doing their job in scripts run in Workbench because the relevant compiler flags were not passed to the `compile` function when the scripts were executed.

This PR parses the given code for \_\_future\_\_  imports and gives the relevant flags to the `compile` function before its output is executed.

**To test:**
The following should run without error, and as described, in Workbench:
```python
# The following line helps with future compatibility with Python 3
# print must now be used [... cut]
closed 2019-10-24 by NickDraper