mantidproject/mantid #27061: Use CMake GenerateExportHeader instead of doing this manually.

martyngigg, 2019-10-22:


DannyHindson, 2020-03-11:
Most of this change involves replacing a DllExport attribute on a class definition with a macro such as MANTID_ALGORITHMS_DLL that is defined in a DllConfig.h file

While doing this I have also done a check on classes that didn't have a DllExport attribute at all. These are classes that probably don't have a unit test defined because without this the xxxxTest executable (eg AlgorithmsTest) can't see the class. I used the following regex to find these:

^class (?!MANTID_ALGORITHMS_DLL).*[^;]$

This finds lines that start with the word "class" that's not followed by MANTID_ALGORITHMS_DLL and finally the line does not end in a semi-colon (so it excludes forward declarations of classes). This matched 4 classes in the Algorithms project: FFTDerivative, PeakFitResult, GravitySANSHelper and Qhelper

DannyHindson, 2020-03-13:
Following this change we should check whether the preprocessor macros such as IN_MANTID_API can be removed. I think they were only used by the manually created dllconfig.h and haev been replaced by the API_EXPORTS macro that is built into cmake:

set_target_properties(API
                      PROPERTIES OUTPUT_NAME
                                 MantidAPI
                                 COMPILE_DEFINITIONS
                                 IN_MANTID_API)

DannyHindson, 2020-03-13:
Similar check on classes without a DllExport attribute in API project (and therefore without unit tests) gives this list:
AlgorithmStartingNotification, AlgoTimeRegister, PartialJacobian, FunctionHandler, TableColumnHelper, TableConstColumnHelper, TableRowHelper, Jacobian, SpectrumInfoIterator, WorkspaceProperty

DannyHindson, 2020-03-19:
Classes without a DllExport attribute in Crystal are: BackgroundStrategy, PeakContainer, DataModeHandler, SCDCalibratePanels
Classes without a DllExport attribute in Crystal are: MaleableCostFunction, CrystalFieldHeatCapacityBase, CrystalFieldMagnetisationBase, CrystalFieldMomentBase, CrystalFieldSusceptibilityBase, RemovePeaks, UserFunction1D, GSLJacobian, JacobianImpl1, HalfComplex, Jacobian, RandomVariateGenerator

DannyHindson, 2021-02-23:
I'm still intending to work on this in one of the next release sprints as a maintenance activity. The task is only partially complete right nw

DanielMurphy22, 2021-08-23:
Think I'll give this another 6 months

DannyHindson, 2022-02-25:
I'll keep this one alive. Still intending to complete at some point - looks like Daniel also had a look at this at some point

DannyHindson, 2022-10-06:
As for earlier chunks of work on this here's a list of classes without a DllExport attribute - the point being that these can't be being unit tested at the moment because the unit test wouldn't have access to the class.
Classes without a DllExport attribute in DataHandling are: BankPulseTimes, FileByteStream (BitStream.h), VectorByteStream (BitStream.h), DetParameters (FindDetectorsPar.h), AvrgDetector (FindDetectorsPar.h), several classes in LoadANSTOHelper.h (ProgressTracker, EventProcessor, EventCounter, EventAssigner, EventAssignerFixedWavelength, FastReadOnlyFile, File), InvalidLogPeriods (LoadEventNexus.h), LoadMuonNexusV2NexusHelper, SpiceXMLNode, MultiPeriodLoadMuonStrategy, ProcessBankData, SinglePeriodLoadMuonStrategy

PR #28335 (2020-03-12): Use cmake generate_mantid_export_header to generate DllConfig.h (Algorithms)
**Description of work.**

One of several pull requests on this topic about attaching a DllExport or DllImport attribute to class definitions. This one makes the change on the header files in the Algorithms project.

The change consists of removing the manually created DllConfig.h header and replacing it with one generated during the build by cmake ie not in GitHub. This PR changes the classes which previously used DllExport over to using MANTID_ALGORITHMS_DLL (a few already used MANTID_ALGORITHMS_DLL from the manually [... cut]
PR #34497 (2022-10-17): Add CMake export header feature to more targets (Part II)
**Description of work.**

Further work on this task. See #34482. This PR extends the work to the Geometry, Kernel, MDAlgorithms and Nexus targets

**To test:**

No functional change so just code review and check the Jenkins jobs all pass

Although lots of files changed, the only interesting edits are to the CMakeLists.txt files. The remaining edits are mainly changes to use the MANTID_\<TargetName\>_DLL macro instead of the DLLExport macro

I've also deleted a .cmake file from the Mantid repository that is actually part of the cmake ins [... cut]