mantidproject/mantid #27065: Developer documentation for Abins

ajjackson, 2019-10-23:
The codebase of Abins is split across an Algorithm and a collection of python modules. Some of the design decisions and requirements for extension/use as an API are not immediately obvious from the code. While work is underway to improve readability of the code, it could also be helpful to provide a page with some specific development comments.

This should include:
- [ ] Preferred units and conventions
- [ ] Instructions for generating/updating system-test data
- [ ] An overview of the data/caching model

ajjackson, 2019-10-30:
From discussion with @AnthonyLim23 it seems that dev-docs will be the best place for this to go. I'm putting together some relevant notes on sampling/broadening as part of work on #27043 so these can form the first contribution to this document.

Harrietbrown, 2021-04-01:
Fixed by #27501

closed 2021-04-01 by Harrietbrown