mantidproject/mantid #27085: Technique tag required in instrument definition

DanielMurphy22, 2019-10-24:
Found by myself and Martyn Gigg

When defining a new instrument in the Facilities.xml file (Under AppData/Roaming/mantidproject/mantid/instrument/, search %APPDATA%)
the instrument definition would not appear in the Mantid GUI unless the technique tag was inserted.

Worth deciding if we actually need to require the technique tag:
- 1) If not, then making it possible for an instrument to be defined without one and for it to appear in Mantid.
OR
-2) If so, then providing an error which explains why the newly defined instrument hasn't been loaded, and telling users to include a technique tag.

Some code used:
```

  <instrument name="electronics">
    <technique>Test Listener</technique>#it won't load the instrument with this line removed
    <livedata>
      <connection name="kafka_event" address="livedata.isis.cclrc.ac.uk:9092" listener="KafkaEventListener" />
    </livedata>
  </instrument>

```

SpencerHowells, 2019-10-24:
@martyngigg We use the technique tag regularly to find indirect or direct instruments when starting a GUI for example.

martyngigg, 2019-10-24:
@SpencerHowells Thanks for the information. Daniel and I just spent ~20 minutes trying to figure out why a new instrument in the `Facilities.xml` wouldn't load. The safest thing is probably to warn that the instrument description was not loaded with a reason to avoid this confusion for the future.

commit cb49d23e (2020-11-26): Log errors with skipping invalid instrument info

When loading Facilities.xml log any exceptions that are raised.

Fixes #27085
commit 5f91830d (2020-11-26): Log errors with skipping invalid instrument info

When loading Facilities.xml log any exceptions that are raised.

Fixes #27085
commit 17a29536 (2020-11-26): Log errors with skipping invalid instrument info

When loading Facilities.xml log any exceptions that are thrown.

Fixes #27085
commit 3097f947 (2020-11-26): Log errors with skipping invalid instrument info

When loading Facilities.xml log any exceptions that are thrown.

Fixes #27085
closed 2020-11-30 by gemmaguest