mantidproject/mantid #27120: Unscripted Testing Mantidplot Group 4

DanielMurphy22, 2019-10-28:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://www.mantidproject.org/Unscripted_Manual_Testing#Groups

DannyHindson, 2019-10-30:
Data Loading - spectrum # filter doesn't work properly on GEM38370_Focussed_Legacy.nxs. Noticed this during induction exercises. Issue already open for this (#11305)

3D graphs - Can't get 3D plot Bars, Scatter menu items to do anything. Help suggests they're only relevant to table workspaces but menu not present for table workspaces. Issue already exists for this (#7894)

Don't have an icat login at present so haven't tested catalog searching & downloading. Will leave this in progress for now for this reason - although note this is also part of group 1 tests

DannyHindson, 2019-10-30:
Closing as suggested by Nick

closed 2019-10-30 by DannyHindson