mantidproject/mantid #27125: Unscripted Testing TOFConverter

DanielMurphy22, 2019-10-28:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
See http://docs.mantidproject.org/v3.12.0/interfaces/TOF%20Converter.html

alicerussell1, 2019-10-29:
Testing on windows.
On workbench a crash is caused by entering 'e . + -'  etc when no numbers surrounding them or by entering a double after `e` e.g. `1e0.5`. This happens on the input value, scattering angle and total flight path.
On mantidplot this gives an error but does not crash.

alicerussell1, 2019-10-29:
This is already an issue #26121

closed 2019-10-29 by alicerussell1