mantidproject/mantid #27131: Unscripted Testing FilterEvents

DanielMurphy22, 2019-10-28:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

NickDraper, 2019-10-29:
testing with windows 10

NickDraper, 2019-10-30:
#27179 created for a crash situation

NickDraper, 2019-10-30:
Ok this interface is pretty fragile, all processing is done in the main thread, and it is too easy to kick off a very big job freezing the interface, but that it the same as previous releases, so passing for now.

closed 2019-10-30 by NickDraper