mantidproject/mantid #27134: Unscripted Testing Engineering Diffraction

DanielMurphy22, 2019-10-28:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: http://docs.mantidproject.org/nightly/interfaces/Engineering%20Diffraction%20Test%20Guide.html

Harrietbrown, 2019-10-29:
This issue is still unresolved #26125

GSAS-II tests are not sufficient to describe the steps to use and test the tab. The [docs](https://docs.mantidproject.org/nightly/interfaces/Engineering%20Diffraction.html#gsas-fitting) for the tab also describe it as being in early stages.

We are currently rebuilding this interface from the scratch as much of the interface has been broken for several releases, with quite a few issue still open from the last run of unscripted testing.

Harrietbrown, 2019-10-29:
#27157 An issue to cover the changes to documentation changes that I will need to make to the tests.
Other currently existing issues that are still replicatable:
#26138, #26115, #20738, #20737

closed 2019-10-30 by Harrietbrown