mantidproject/mantid #27136: Unscripted Testing Workspace Groups

DanielMurphy22, 2019-10-28:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Follow the examples at http://docs.mantidproject.org/nightly/concepts/WorkspaceGroup.html 
* Ensure MantidPlot displays Group Workspaces correctly in the workspace tree

DavidFair, 2019-10-30:
Testing is partially completed - once #27175 is fixed I can re-test

DavidFair, 2019-10-30:
Yes, I ended up re-testing on a different machine

closed 2019-10-30 by DavidFair