mantidproject/mantid #27140: Unscripted Testing ISIS SANS (v2 GUI)

DanielMurphy22, 2019-10-28:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

# SANS GUI Testing

<div class="contents" data-local="">

</div>

## Data reduction

*Preparation*

  - Get the ISIS sample data from the website
  - Ensure that the ISIS sample data directory is in the search
    directories

**Time required 20 - 30 minutes**

-----

1.  Open `Interfaces` \> `SANS` \> `SANS v2`

2.  Set `Instrument` to `LOQ`

3.  Choose `Load User File`; from the ISIS sample data in the `loqdemo`
    folder, choose `Maskfile.txt`

4.  Choose `Load Batch File`; from the ISIS sample data in the
    <span class="title-ref">loqdemo</span> folder, choose
    `batch_mode_reduction.csv`

5.    - In the `Settings` tab:
        
          - Go to `Mask`
          - Click `Display Mask`
          - This should give an instrument view with a circle at the
            centre

6.    - In the `Runs` tab
        
          - Click `Process`
          - After some seconds the rows should turn green
          - In the Main window there should be a series of new
            workspaces; 3 group workspaces and 4 2D workspaces
          - Change the first column of the first row to 74045; click
            process
          - The row should turn blue; hovering over the row should give
            an error message
          - Change the `Reduction` button to 2D
          - Check the `Plot Results` box
          - Click `Process`
          - A plot window will open; initially empty, then with a red
            line
          - Change `Reduction` back to 1D
          - Click `Process`
          - In the plot you should end up with a red and a black line
            plotted
          - Check the `Multi-period` box
          - 6 additional rows should appear in the table
          - Check that the `Add`, `Delete`, `Copy`, `Paste`, `Cut` and
            `Erase` icons work as expected
          - Delete all rows and re-load the batch file as previously

7.    - In the `Settings` tab
        
          - Set `Reduction Type` to `Merged`
          - Return to the `Runs` tab
          - Ensure `Plot results` is checked
          - Click `Process`
          - This should result in a plot with three lines

8.    - In the `Beam centre` tab
        
          - Click
          - A plot should appear after some seconds, with 4 lines
          - The 4 lines should gradually get closer together
          - This will run for some time, probably minutes

9.    - In the `Sum Runs` tab
        
          - Enter `74044, 74019` in the top line
          - Enter `LOQ74044-add` as Output file
          - Ensure that the `Output directory` is in you r managed paths
          - Click `Sum` at the bottom
          - Go back to the `Runs` tab
          - Remove all rows
          - Reload the batch file as before
          - Change the first column of both rows to `LOQ74044-add`
          - Click `Process`
          - This should now process as before

10.   - In the `Diagnostic Page` tab
        
          - For run choose `Browse` and load the `LOQ74044.nxs` file
          - Click each of the `Integral` buttons
          - They should produce plots
          - Check the `Apply Mask` boxes and click the buttons again
          - They should produce new, slightly different plots

11.   - In the `Runs` tab
        
          - Check all variables and inputs for tool tips by hovering
            over them

RichardWaiteSTFC, 2019-10-29:
Waiting on PR #27038 and #27040 to test. But some of the instructions for the testing are inaccurate/can be clarified:
(1) There are two ISIS SANS interfaces (ISIS SANS and Old ISIS SANS) not SANS v2
(6) 
- Replace Process with process All.
- There are actually 4 of each type of workspace (when both are processed, see screenshot))
- 6 additional COLUMNS  appear in the table when multi-period is checked
(7) Settings > general, Scale, Event Slice , Sample > Reduction Type to Merged
(8) Click Run

[image]

gemmaguest, 2019-10-31:
The v2 GUI has been renamed to just `ISIS SANS` now. I've updated the instructions to distinguish it from the testing for the old GUI in #27126

RichardWaiteSTFC, 2019-10-31:
In the nightly build I can't load data. When I try to process a run or display the mask I get the following error  in the results log (though there is no indication in the interface itself, just the display mask button is faded and says Processing...).
[image]
However I have the data directory with the file LOQ74044.nxs at the top of the list of directories mantid searches through and have set LOQ to be my default instrument. I follow the same steps in Mantid 4.1 and it works.

RichardWaiteSTFC, 2019-10-31:
SANS interface passes all tests but for known issue that the data needs to be loaded twice before processing, otherwise it produces the error in my previous comment.

closed 2019-10-31 by RichardWaiteSTFC