mantidproject/mantid #27141: Unscripted Testing Live Data ISIS

DanielMurphy22, 2019-10-28:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://www.mantidproject.org/MBC_Live_Data_Simple_Examples

gemmaguest, 2019-10-31:
Tested on Windows 10 with the nightly version 4.1.20191030.2045

### New issues
I found these new issues in Workbench when following the unscripted testing instructions when I got to the final example of ADARA_FakeEvent with post-processing. I think it may be to do with AccumulationMethod=Add. Note that the test instructions have disappeared from the wiki but you can see them on PR #27218

- Workbench hung when I plotted the spectra for the output live and accumulation workspaces. The hang happened at around chunk 15 the first time but not till chunk 410 a second time. This seems ok in Workbench v4.1.0 (at least till around chunk 600, although I had another issue where one of the plots went blank) and is ok in MantidPlot in either version. 
- Another time doing the same thing, both of the output workspaces disappeared even though live data was still running. I couldn't reproduce this.
- The plots don't rescale as the workspaces update, so you see the graph disappear off the top of the plot window as the values increase. They used to rescale in v4.1.0.

This is a screenshot of the hang, where you can also see one plot has disappeared and the other is going that way:
[image]

### Existing issues
- In Workbench, FakeISISEventDAE doesn't seem to start first time round. Nothing shows in the Algorithm Progress dialog after running it. However, it does appear after running a second time but with progress `Waiting for client - 0%` (rather than `Fake DAE - 0%` for FakeISISHistoDAE). This is the same as in v4.1.0. (In MantidPlot it appears first time round but the progress is empty.)

gemmaguest, 2019-10-31:
@NickDraper I'll create issues for the problems I found in the morning so let me know if there's anything you're already aware of

gemmaguest, 2019-11-01:
Closing as I've created issues for the problems found

closed 2019-11-01 by gemmaguest