mantidproject/mantid #27143: Unscripted Testing M-slice

DanielMurphy22, 2019-10-28:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://mantidproject.github.io/mslice/ and follow Quick Start, this should probably be tested in iDAaaS

hsautessella, 2019-10-29:
A few issues found, I don't know that any are important for the upcoming release:

- Uncaught exceptions when loading some files:
    - Attempt to load MUSR00015196.nxs (in the Mantid unit test data).
    - The following AttributeError is raised:
```
Traceback (most recent call last):
      File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\widgets\dataloader\dataloader.py", line 55, in activated
        self.load(False)
      File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\widgets\dataloader\dataloader.py", line 104, in load
        self._presenter.load_workspace(self.get_selected_file_paths(), merge)
      File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\presenters\data_loader_presenter.py", line 35, in load_workspace
        self._load_ws(file_paths, ws_names, force_overwrite)
      File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\presenters\data_loader_presenter.py", line 51, in _load_ws
        load(filename=file_paths[i], output_workspace=ws_name)
      File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\models\workspacemanager\workspace_algorithms.py", line 163, in load
        workspace.e_mode = get_EMode(workspace.raw_ws)
    AttributeError: 'Load_returns' object has no attribute 'raw_ws'
```
- When you click refresh in the Data Loading tab, the "Date Modified" and "Size" column switch around.
- The text does not fit inside the "Save to Workbench" button on the MD Histo tab

[image]

- Error reporter pop-up when trying to load a GroupWorkspace (e.g. LARMOR00003368 in unit test data).
- Braces not closed on plot units when setting the intensity to Chi''(Q, E) Magnetic

[image]

- The print button on the plot toolbar did not seem to do anything

- When saving an image, the default format in the file window is .nxs

- If you plot a slice and add some information (e.g. Bragg peaks), then you change the colormap, the Bragg peaks will disappear, but the Bragg peaks will still be checked in the "Information" menu.

- Deleting a workspace in the main Workbench GUI does not remove the workspace from MSlice. Clicking the deleted workspace in MSlice raises the error dialog.

- Sometimes when you attempt to load a file, an error message will be written to the bottom of the MSlice window saying "MSlice was not able to load the selected file". However, the file is loaded into the Workspace Manger tab, but you cannot perform any Powder/Slice/Cut operations on it.

closed 2019-10-29 by hsautessella

hsautessella, 2019-10-29:
Tested on Windows