mantidproject/mantid #27149: Indirect reduction > remove automatic suffixes

gvardany, 2019-10-28:
Indirect ILL reduction automatically appends suffixes to the workspace name after what is requested by the caller. This was originally implemented to accomodate better the requirements of the indirect interfaces. But running from scripts this does not make much sense, moreover, creates more confusion.
There should be a possibility to switch this behaviour off, without breaking the backwards compatibility.

SpencerHowells, 2019-10-28:
@gvardany the interfaces have a tool for switching the option on/off so anything new should be compatible with that

gvardany, 2019-10-29:
Yes, exactly. That's why we can now offer an option not to append anything in the algorithms either.

closed 2021-03-02 by stale[bot]