mantidproject/mantid #27159: Instrument view causes MantidPlot to crash while using MantidPlot over a remote desktop connection

StephenSmith25, 2019-10-29:
### Issue
When trying to use the instrument viewer while running MantidPlot over a remote desktop connection Mantid plot crashes. This is repeatable on both the latest nightly build, and also the 4.1 release. 

## Error
### Event Viewer
Faulting application name: MantidPlot.exe, version: 0.0.0.0, time stamp: 0x5db3a1e5
Faulting module name: atig6pxx.dll, version: 23.20.806.2048, time stamp: 0x5a5e1e1b.
Exception code: 0xc0000005
Fault offset: 0x0000000000009e54
Faulting process id: 0x4ec8
Faulting application start time: 0x01d58e49d2e19169
Faulting application path: C:\MantidNightlyInstall\bin\MantidPlot.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\c0323553.inf_amd64_8c35f7d45fc826dd\atig6pxx.dll
Report Id: 0c4a3933-781b-4989-8804-337144f66bce
Faulting package full name: 
Faulting package-relative application ID:

### Platforms affected
Windows 10 - The problem appears to be system dependent though.

DanielMurphy22, 2019-11-01:
Seems like the same issue so referencing: #27064

DanielMurphy22, 2020-03-03:
I can reproduce using my macbook pro 2017 on Catalina remoting into my Windows 10 desktop

martyngigg, 2020-03-18:
Yes this has been an issue. You can disable OpenGL by adding `MantidOptions.InstrumentView.UseOpenGL = Off` to the properties file. There is an option in preferences I think.

martyngigg, 2020-03-18:
It's annoyingly difficult to determine whether OpenGL is going to work or not but there could be better checks we can add.

closed 2021-03-02 by stale[bot]