mantidproject/mantid #27162: ComputeIncoherentDOS algorithm causes docs-html build in debug on windows

Harrietbrown, 2019-10-29:
### Expected behavior
All Documentation should build correctly in visual studios for both release and debug
### Actual behavior
In debug mode `ComputeIncoherentDOS-v1.rst` and `DirecttoolsPythonModule.rst` produce an error in debug mode.
Both these builds fall over when trying to use `ComputeIncoherentDOS`. in line 86 in `ComputeIncoherentDOS-v1.rst` and in line 132 and 133 in `DirecttoolsPythonModule.rst`
### Steps to reproduce the behavior
1. Open the vidual studios project for a master build of mantid.
1. in debug mode, attempt to build docs-html object.
### Platforms affected
Windows

gvardany, 2019-11-06:
I started to see `ComputeIncoherentDOS-v1.rst` and `DirectILLTubeBackground-v1.rst` segfault also on release build on mac. To be investigated...

DannyHindson, 2020-02-26:
I noticed this problem yesterday while doing something else and found this issue while wondering if it had come up before.

I had a look into it and the problem is that Rebin2D assumes the input workspace has histograms with an xmode of BinEdges rather than Points. ComputerIncoherentDOS transposes a 2D workspace before calling Rebin2D on it and this results in a workspace with an xmode of Points being supplied. One of the loops in Rebin2D reads beyond the end of the x vector causing an index out of bounds error that's only raised in debug mode.

I think this can be fixed by replacing two calls to inputWS->x(i) with inputWS->binEdges(i) in Rebin2D.cpp (L87) and FractionalBinning.cpp (L611)

I'll do this as a maintenance change if I get chance

closed 2021-03-02 by stale[bot]