mantidproject/mantid #27164: Scripts to install GSAS-II is not fuctioning on windows

Harrietbrown, 2019-10-29:
### Expected behavior
The instalation instructions for GSAS-II in the mantid package given in: https://docs.mantidproject.org/nightly/algorithms/GSASIIRefineFitPeaks-v1.html#installing-gsasii should be usable for all platforms
### Actual behavior
changes in the api for pip has made the windows installer non functional
### Steps to reproduce the behavior
From the mantid source folder, run: `scripts\GSAS-II\install_gsas_vetted.bat`
### Platforms affected
Windows

DanielMurphy22, 2021-02-17:
Following the normal GSAS isntall instructions for Windows work, but it doesn't quite work with Mantid for [algorithm GSASIIRefineFitPeaks](https://docs.mantidproject.org/v3.7.1/algorithms/GSASIIRefineFitPeaks-v1.html).

Also [installation on Linux](https://subversion.xray.aps.anl.gov/trac/pyGSAS/wiki/InstallLinux) is difficult due to problems pip installing [wxPython](https://wxpython.org/blog/2017-08-17-builds-for-linux-with-pip/index.html). Worth checking with IDAaaS on how they do this.

DannyHindson, 2021-11-23:
The investigation around this is now complete.

Problems were possibly caused by python mismatch between GSAS II and Mantid (now on python 3.x)

We believe that trying to install a specific version of GSAS within the Mantid installation directory is going to be difficult to make robust across all platforms including IDAaaS. So we will assume the user has installed GSAS-II separately.

Will create separate issues to take the ENGINX GSAS II integration work forward

closed 2021-11-23 by DannyHindson