mantidproject/mantid #27221: Muon AlC interface crashes

alicerussell1, 2019-11-01:
### Expected behavior
If you enter the first set of data, tick `Auto` and press `Load` it should load the data

### Actual behavior
Hard crash

### Steps to reproduce the behavior
Set the default instrument to EMU, and turn on the data archive
`Interfaces`->`Muon`->`ALC`
Enter some valid data for the first run and tick the `Auto` box
Press `Load`
It crashes

### Platforms affected
Tested on windows but probably all

alicerussell1, 2019-11-04:
I had another look at this since you couldn't reproduce it, it seems that it only crashes when I have `//emu/data/` in manage user directories. It also causes a crash in the previous release.

StephenSmith25, 2019-11-04:
I tried the same on Ubuntu 18.04, and added the //emu/data folder to the user directories. It also causes Mantid to crash, so this appears to be the issue. I'll look into it.

PR #27264 (2019-11-12): Muon AlC interface crashes
**Description of work.**
The ALCLatestFileFinder expected the instrument name to be case-consistent, e.g EMU, or emu. If it was supplied with a file with an upper case detector name, e.g EMU00066666 from //emu/data it would crash. The crash was due to querying an empty vector, which now has a return check, on the off chance the underlying directory changes (e.g wiped). The instrument check has also been changed so that it is now case-insensitive.

**To test:**
To replicate the original issue

    Set the default instrument to EMU, and turn off the data archive
    Ad [... cut]
closed 2019-11-12 by StephenSmith25