mantidproject/mantid #27225: Workbench can hang with plots of live data

gemmaguest, 2019-11-01:
### Expected behavior
Workbench should not hang

### Actual behavior
Workbench sometimes hangs when showing plots of live data when live data is still accumulating.

### Steps to reproduce the behavior

I found these new issues in Workbench when following the unscripted testing instructions when I got to the final example of ADARA_FakeEvent with post-processing. I think it may be to do with AccumulationMethod=Add. Note that the test instructions have disappeared from the wiki but you can see them on PR #27218

- Workbench hung when I plotted the spectra for the output live and accumulation workspaces. The hang happened at around chunk 15 the first time but not till chunk 410 a second time. This seems ok in Workbench v4.1.0 (at least till around chunk 1000, although I had another issue where one of the plots went blank) and is ok in MantidPlot in either version. 
- Another time doing the same thing, both of the output workspaces disappeared even though live data was still running. I couldn't reproduce this.

[image]

### Platforms affected
Seen on Windows 10

DannyHindson, 2019-11-05:
Quick update on this. I've spent some time this afternoon trying to recreate this problem by following the live data test script and I've not been able to. I have been using the nightly build (mantidnightly-4.1.20191104.1440-win64.exe)

I've done several runs where I've waited for the chunk to get up to ~1000 and Workbench hasn't hung for me. I'll have another look tomorrow using a local build in case that was used (I remember the Windows nightly build wasn't working for a couple of days last week)

I do see the other issue that Gemma mentions from her unscripted testing where the plots don't resize and also where the  FakeISISEventDAE algorithm doesn't appear to start properly.

gemmaguest, 2019-11-06:
@DannyHindson I can't reproduce this any more so I'm inclined to close it if you still haven't seen the problem?

DannyHindson, 2019-11-06:
I've not been able to reproduce it either after further tries this morning.

I noticed while waiting that the messages window shows an occasional feature usage message (from LoadLiveData.v1) if you have the logging level set to Debug. This seems to occur every minute or two:

Sending https POST request to: https://reports.mantidproject.org/api/feature

There's also a project recovery process running every minute that shows up in the same window.

Is it possible one of these kind of periodic processes might under different circumstances cause a hang? Clutching at straws a bit but thought I'd mention it. Neither process resulted in a hang for me.

I'm happy to close if this doesn't sound v likely.

gemmaguest, 2019-11-06:
It looks like the first one of those is usage reporting (we should actually turn that off as developers or it pollutes the usage results!) and the second is project recovery, neither of which I would suspect of causing a problem. I'll close this because I don't think there's much more we can usefully do for now.

closed 2019-11-06 by gemmaguest