mantidproject/mantid #27228: Workbench plots don't rescale when live data updates

gemmaguest, 2019-11-01:
### Expected behavior
If the data in the workspace changes when running live data, the behaviour in workbench v4.1.0 and in the current MantidPlot was that any open plots would rescale to show all of the data, so I assume they should still do this.

### Actual behavior
The plots do not rescale

### Steps to reproduce the behavior
I found this in the Workbench nightly 4.1.20191030.2045 when following the unscripted testing instructions when I got to the final example of ADARA_FakeEvent with post-processing where AccumulationMethod=Add. Note that the test instructions have disappeared from the wiki but you can see them on PR #27218.
- Follow the test instructions for the setup and the final test case.
- Plot the live and accumulation workspaces while live data is still running.
- See that the plots do not rescale

### Platforms affected
Seen on Windows 10. Probably all.

DavidFair, 2019-11-05:
It appears that the changes in #25985 regressed this

Talking to @Mantid-Matthew the Muon GUI should not scale, however all other plots should as this has been the default for MantidPlot prior

Harrietbrown, 2019-11-11:
I'm trying to recreate the issue, but the live output following th wiki instructions is 0 for all points so I can't tell if the plot is updating. I might have done something wrong setting it up, is it supposed to be all 0?

gemmaguest, 2019-11-11:
They shouldn't be all 0 - I get plots like this:
[image]

I had a problem where I was seeing all or mostly 0's for a while but I think this was because I hadn't noticed the update interval was unset, which I think means you get a 60 second interval by default. I think you get 0's until it refreshes after the first 60 seconds (I'm not sure that's desired behaviour but is a known "feature"!).

Harrietbrown, 2019-11-12:
I've got my update every set to 1 second. with more than one spectra the values are between 0 and 1 but are all constant

Harrietbrown, 2019-11-14:
I couldn't find an underlying cause, but I suspect it might be similar to what we see when you plot a workspacem then load a different workspace but with the same name. The line updates but the axis doesn't rescale.

martyngigg, 2020-10-27:
So I think this has been fixed for the line plots. The Y-scale increases over time as the data accumulates.

For the colorfill plot of the accumulated workspace though I get an error during update:
```
Error occurred in handler:
Traceback (most recent call last):
  File "/home/mgigg/Code/mantidproject/mantid/qt/applications/workbench/workbench/plotting/figuremanager.py", line 51, in wrapper
    func(*args, **kwargs)
  File "/home/mgigg/Code/mantidproject/mantid/qt/applications/workbench/workbench/plotting/figuremanager.py", line 124, in replaceHandle
    redraw_this = ax.replace_workspace_artists(workspace)
  File "/home/mgigg/Code/mantidproject/mantid/Framework/PythonInterface/mantid/plots/mantidaxes.py", line 425, in replace_workspace_artists
    is_empty_list = [workspace_artist.replace_data(workspace) for workspace_artist in artist_info]
  File "/home/mgigg/Code/mantidproject/mantid/Framework/PythonInterface/mantid/plots/mantidaxes.py", line 425, in <listcomp>
    is_empty_list = [workspace_artist.replace_data(workspace) for workspace_artist in artist_info]
  File "/home/mgigg/Code/mantidproject/mantid/Framework/PythonInterface/mantid/plots/mantidaxes.py", line 1540, in replace_data
    new_artists = self._data_replace_cb(self._artists, workspace)
  File "/home/mgigg/Code/mantidproject/mantid/Framework/PythonInterface/mantid/plots/mantidaxes.py", line 867, in _update_data
    return self._redraw_colorplot(plotfunctions_func, artists, workspace, **kwargs)
  File "/home/mgigg/Code/mantidproject/mantid/Framework/PythonInterface/mantid/plots/mantidaxes.py", line 902, in _redraw_colorplot
    artists_new = colorfunc(self, workspace, norm=artist_orig.norm,  **kwargs)
TypeError: imshow() got multiple values for keyword argument 'norm'
```

martyngigg, 2020-10-27:
To get the ADARA example to work I had to use rebin parameters `6e3,500,20e3` as it said the data was between 6000 & 23000 microseconds.

PR #30046 (2020-12-08): Fix handling image replotting when the underlying workspace updates
**Description of work.**

Fixes a bug where an exception would be shown the log when a workspace backing a colorfill plot was replaced.

Further tests have been added to flex this area of plotting.

**To test:**

* Load any workspace, e.g. `<builddir>/ExternalData/Testing/Data/SystemTest/irs26173_diffspec_red.nxs`
* Run any algorithm to replace the workspace, e.g. Scale with a factor of 1000
* The image should update and not write an error to the log.

Fixes #27228 

---

#### Reviewer ####

Please comment on the following ([f [... cut]
closed 2020-12-08 by gvardany