mantidproject/mantid #27244: Instrument View window shrink

DanielMurphy22, 2019-11-04:
### Expected behavior

### Actual behavior

### Steps to reproduce the behavior

### Platforms affected

DanielMurphy22, 2020-02-11:
This is not an issue on MantidPlot

DanielMurphy22, 2020-02-11:
Interestingly when I connected a second screen to the Macbook, this issue was not longer a problem on the second screen, only on the main laptop screen!

DanielMurphy22, 2020-02-11:
Also, a related issue about peak numbers appearing "Blurry" in workbench has been reported. The user reported this error message being output which may be relevant to either (or neither) of these issues: 
`Attribute Qt::AA_EnableHighDpiScaling must be set before QCoreApplication is created.`

sf1919, 2020-09-09:
I've been trying to follow the Basic Introductory training and in doing the Exercises for Instrument View (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/06_exercises_c.html
) I'm having the same issue with the plot. I'm on Windows 10 rather than a Mac though. For ref this is the Nightly download of version 5.0 from 07/09/20.

The issue also means I'm unable to complete the exercise as the mini plot doesn't work and remains blank for all the detectors.

AnthonyLim23, 2020-09-09:
@gemmaguest said its ok on Linux

gemmaguest, 2020-09-09:
> 
> 
> @gemmaguest said its ok on Linux

With the caveat that I'm using X11 forwarding to display locally on my Windows machine... which has affected a few display-related things in the past....

gemmaguest, 2020-09-15:
Fixed by #29432

PR #29432 (2020-09-15): Fix OpenGL viewport size for 4K resolutions
**Description of work.**

**To test:**

* You will need a monitor connect with a 4K resolution
* Build and start workbench
* Open the instrument view on any instrument and flick between the projections -> each should fill the full widget area

Fixes #27244

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Stand [... cut]
closed 2020-09-15 by gemmaguest