mantidproject/mantid #27247: Fit Function Window gets Stuck Popped Out in "I(Q, t) Fit" tab on Workbench

ConorMFinn, 2019-11-04:
### Expected behavior
The widget cannot be popped out on workbench. 
### Actual behavior
Once the widget has been popped out of the main interface, it cannot be moved and clicking the button again (which would return the widget to the main GUI in plot) does nothing.
### Steps to reproduce the behavior
1. Open Workbench
1. `Interfaces` -> `Indirect`-> `Data Analysis`
3. `I(Q, t) Fit` tab
4. Click the separate window button in the top right corner of the fit function widget.
### Platforms affected
Button only appears on

SpencerHowells, 2019-11-04:
@ConorMFinn If I remember correctly this is a general problem on the Workbench with docking. The miniPlot widget appears to have it disabled ie no icon. I think we decided to temporarily disable it on all tab - see eg ConvFit tab.

SpencerHowells, 2019-11-04:
@ConorMFinn Using Ubuntu the undocked widget does not move around. This is a bug in Qt.

Mantid-Matthew, 2019-11-06:
This was in fact a generic issue with all the tabs when the IndirectDataAnalysis GUI was ported to workbench. The decision was made to remove the ability to pop out the sub widgets for now which is what should be done here.

gemmaguest, 2019-11-07:
Fixed by #27296

PR #27296 (2019-11-07): Add check on qt version to disable docking in I(Q, t) fit tab browser
This PR disables the ability to pop out the fit browser in `Indirect` -> `Data Analysis` in workbench.
This implements the same QT version check to `IndirectFitAnalysisTab.cpp` that exists in `IndirectFitAnalysisTabLegacy.cpp`

**To test:**
1. open workbench
1. `Interfaces` -> `Indirect`-> `Data Analysis`
1. I(Q, t) Fit tab
1. The button to pop the Fit Function should not be available.
1. do the same with MantidPlot, the button should be present and functional.

Fixes #27247

*This does not require release notes* because thi [... cut]
closed 2019-11-07 by gemmaguest