mantidproject/mantid #27289: Workbench markers cannot be deleted

AnthonyLim23, 2019-11-06:
Open workbench and make a plot
Add a marker (vertical or horizontal)
The only way to delete it is by right clicking and picking delete 

We would like a delete option in the edit menu (probably best to just close the edit window afterwards)

martyngigg, 2019-11-07:
This feels like a feature request. Is it really necessary for the release?

martyngigg, 2019-11-07:
The title is also a little misleading as they can be deleted but we'd like something a little more convenient?

NickDraper, 2019-11-08:
Pushing out of the release as we will not have time to do this, and there is a work around

NickDraper, 2020-02-04:
I've looked into this and I'm going to close it as a won't fix for the following reasons:
1. To get to the Edit dialog you have to move your mouse past the delete option on the context menu
1. It is not always a good thing to have multiple ways of doing the same thing
1. Within the code the dialog does not have a link to the FigureInteraction.py that needs to remove references to the marker.   You could link it up with a signal and slot, but given the other two points it seems unnecessary.

closed 2020-02-04 by NickDraper